Memotech-Bill/PicoBB #10: PIN I/O

georgestheking, 2022-09-06:
Hi,

I find a program "blink" to SET / RESET pins.
What is the syntax to read a PIN STATE ?

Thanks for all

Georges

cuedcad, 2022-09-09:
If you want the status of a single pin, try SYS"gpio_get",pin% TO status%:P.status%

For higher speed on repetative calls can try G%=SYS"gpio_get":A%=pin%:S%=USRG%
if the pin number i.e. A% is not changing then just need S%=USRG% once set.
if you need the pin% to change it can be made faster using an assembler routine

BW

Memotech-Bill, 2022-09-10:
blink.bbc is an old program, written before the BASIC command SYS supported the Pico SDK. gpio.c illustrates using SYS to blink the LED, and is the example to follow.

The previous reply shows you how to use SYS to get the GPIO status.

closed 2023-04-15 by Memotech-Bill