Memotech-Bill/PicoBB #11: Error about "**multiple definition of `libtop'**" when compiling on Ubuntu 22.04 amd64

CHiPs44, 2022-09-27:
I get an error about "**multiple definition of `libtop'**" when I try to make the project with Ubuntu 22.04 on an Intel 64 bits CPU :

```
~/src/PicoBB/bin/pico$ export PICO_SDK_PATH=/home/chips/src/pico-sdk/
~/src/PicoBB/bin/pico$ export PICO_EXTRAS_PATH=/home/chips/src/pico-extras/
~/src/PicoBB/bin/pico$ export PICO_BOARD=vgaboard_sd
~/src/PicoBB/bin/pico$ LANG=C make
rm -rf build
mkdir -p build
cd ../../src/pico && symbols/header_funcs.py /home/chips/src/PicoBB/bin/pico//build/symbols_sdk.txt /home/chips/src/PicoBB/bin/pico//build/pico_stub.c \
	../../src/pico/symbols/sdk_headers.txt ../../src/pico/symbols/sdk_cyw43_gpio.txt
../../src/pico/symbols/merge_funcs.py build/sympico.h build/symbols_sdk.txt
cd build && PICO_SDK_PATH=/home/chips/src/pico-sdk/ \
	cmake -DPICO_BOARD=vgaboard_sd_w -DSTDIO=PICO -DLFS=Y -DFAT=Y -DSOUND=I2S \
		-DSTACK_CHECK=4 -DMIN_STACK=Y -DPRINTER=N -DSERIAL_DEV=0 \
		-DCYW43=GPIO -DBBC_SRC=../../BBCSDL --no-warn-unused-cli -S ../../../src/pico -B .
Not searching for unused variables given on the command line.
Using PICO_SDK_PATH from environment ('/home/chips/src/pico-sdk/')
PICO_SDK_PATH is /home/chips/src/pico-sdk
Defaulting PICO_PLATFORM to rp2040 since not specified.
Defaulting PICO platform compiler to pico_arm_gcc since not specified.
-- Defaulting build type to 'Release' since not specified.
PICO compiler is pico_arm_gcc
Using PICO_EXTRAS_PATH from environment ('/home/chips/src/pico-extras/')
-- The C compiler identification is GNU 10.3.1
-- The CXX compiler identification is GNU 10.3.1
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/arm-none-eabi-gcc
Build type is Release
Using PICO_BOARD from environment ('vgaboard_sd')
...
[100%] Linking CXX executable bbcbasic.elf
/usr/lib/gcc/arm-none-eabi/10.3.1/../../../arm-none-eabi/bin/ld: CMakeFiles/bbcbasic.dir/home/chips/src/PicoBB/src/bbpico.c.obj:/home/chips/src/PicoBB/src/bbpico.c:340: multiple definition of `libtop'; CMakeFiles/bbcbasic.dir/home/chips/src/PicoBB/BBCSDL/src/bbmain.c.obj:/home/chips/src/PicoBB/BBCSDL/src/bbmain.c:50: first defined here
collect2: error: ld returned 1 exit status
make[3]: *** [CMakeFiles/bbcbasic.dir/build.make:1767: bbcbasic.elf] Error 1
make[3]: Leaving directory '/home/chips/src/PicoBB/bin/pico/build'
make[2]: *** [CMakeFiles/Makefile2:1744: CMakeFiles/bbcbasic.dir/all] Error 2
make[2]: Leaving directory '/home/chips/src/PicoBB/bin/pico/build'
make[1]: *** [Makefile:91: all] Error 2
make[1]: Leaving directory '/home/chips/src/PicoBB/bin/pico/build'
make: *** [../../src/pico/Makefile:179: bbcbasic_pkg.uf2] Error 2
```

CHiPs44, 2022-09-28:
It seems to come either from latest pico-sdk and/or pico-extras.

I have similar problems with compiling the pt52 project from DarkElvenAngel (https://gitlab.com/DarkElvenAngel/pt52), but latest pico-playground generates all UF2 files flawlessly.

fred260571, 2023-12-06:
Same problem here with latest pico-playground and pico-extras on MacOSX

fred260571, 2023-12-06:
Solved the problem by commenting the lines at /BBCSDL/src/bbmain.c  at line 49

```
#ifdef PICO
void *libtop ;          // For stack overflow checking
#endif
```

closed 2025-09-14 by Memotech-Bill