Memotech-Bill/PicoBB #12: TIME$ on raspberry pico

nbrok, 2023-03-10:
If i try to set the time with TIME$="FR.10 Mar 2023,17:00:00" I get the message mistake.
If it needs a RTC on which GPIO pins does it expect the I2C RTC?

Memotech-Bill, 2023-03-10:
PicoBB is derived from BBCSDL. BBCSDL does not support setting the system clock (see [here](https://www.bbcbasic.co.uk/bbcwin/manual/bbcwin7.html#times), therefore PicoBB does not have that capability either.

Since the Pico does have inbuild support for a RTC I will consider adding support for that, but there is always a trade-off between added features and memory available for user programs.

Since most of the C SDK functions are available through the BASIC SYS statement you can always provide your own implementation of the internal or an external RTC.

nbrok, 2023-03-10:
The link you gave tells something else, Perhaps copied from the original BBC basic manual. So a comment in that manual that BBCSDL doesn't support setting the time$ would be helpful. But I appreciate that you are considering to add rtc functionality.

Memotech-Bill, 2023-03-11:
From the BBC BASIC Manual:

"Because the system clock is a shared resource, and changing it could adversely affect other programs, **you can only assign a new value to TIME$ in BBC BASIC for Windows**, and only if you have the appropriate administrative privileges."

So you cannot assign a new value to TIME$ in BBCSDL.

nbrok, 2023-03-16:
Thanks for adding RTC  in the newest version, now I can set the internal clock.

closed 2023-03-24 by Memotech-Bill