Memotech-Bill/PicoBB #23: suggestion: add keyboard support on picocom but allow display as well

DuckyPolice, 2024-05-02:
i want to be able to to use my computer with picocom with serial interface but use a little oled or lcd. how would i do this?

DuckyPolice, 2024-05-02:
or with a VGA USB sheild (still dont know which one is good price and works good and audio would be nice so if you have recommendation that would be lovely)

Memotech-Bill, 2024-05-03:
There is currently no built-in support for OLED or LCD displays. For simple text LCD displays you may be able to use the SDK GPIO routines called via SYS.

The option for USB or Serial console with VGA display already exists, see https://github.com/Memotech-Bill/PicoBB#console-version-with-vga-graphics

The only board tested with VGA and sound output is https://shop.pimoroni.com/products/pimoroni-pico-vga-demo-base

DuckyPolice, 2024-05-04:
ok, after learning about this project a bit more and understanding how inputs and outputs work in it, i understand that you can do that. also, quick question but is there a good place to find bas files or source code i could paste into to play games or use programs? like an archive of programs/games that work here with catagorisations of ascii and display (since the vga board is out of stock and i cant afford it currently because i am 14 and don't have a job until summer.) like where did you find all of the programs that are inside and can be found with "*DIR"? also a command to turn on lowercase syntax would be nice to so you dont have to constantly change capslock to run 

`LOAD "snake"`

or something like that you could just do

`load "snake"`

sorry if i am a bit hard to understand i am autistic so its hard to communicate in a way that is understandable sometimes

DuckyPolice, 2024-05-04:
(ascii space invaders or tetris would be fun btw)

Memotech-Bill, 2024-05-05:
The best documentation for BBC BASIC is the [BBCSDL Manual](https://www.bbcbasic.co.uk/bbcwin/manual/). Note that it is written for the PC & Linux version, so some of it, particularly chapter 2, is not relevant to the Pico version. But the BASIC language and many of the star commands are applicable. See [my notes](https://memotech-bill.github.io/PicoBB/) for some of the differences (also in docs/index.html).

The command [*lowercase](https://www.bbcbasic.co.uk/bbcwin/manual/bbcwin8.html#lowercase) can be used to enable entering commands in lower case.

I am not aware of any single collection of BBC BASIC programs. There are many people who use BBC BASIC:

* [People](https://stardot.org.uk/) who still use one of the original BBC micros or Archimedes.
* People who use any of the other retro versions (CP/M, Spectrum,...).
* People who use Richard Russel's [Windows or SDL versions](https://www.bbcbasic.co.uk/bbcbasic.html).
* People who use [Brandy Basic](https://jaguar.orpheusweb.co.uk/branpage.html).
* Others...

You will have to look around. You will also need to take account of the limited size of the Pico. You may be best off looking for programs written for the original Micro. You may still need to adapt them slightly, a good learning experience.

The VGA output actually only requires a few resistors on some of the GPIO pins, see the circuit diagram in [Hardware design with RP2040](https://datasheets.raspberrypi.org/rp2040/hardware-design-with-rp2040.pdf). Sound is a bit more difficult, although for PWM sound you could just use resistor + capacitor filters on the GPIO pins without the buffer chip used in the Hardware Design.

DuckyPolice, 2024-05-10:
could you build the vga + serial one for me? im having trouble building it. maybe add it in releases or something. for me make is pretty broken idk why

Memotech-Bill, 2024-05-13:
I have made a couple of patches, and updated the instructions for building a version with USB console plus VGA display.

If you follow the updated instructions it will build a UF2 file. I do not currently have a Pico set up to actually test the resulting file.

If you want VGA plus UART console, then you probably need to create your own board definition file to define which pins to use. The parameters for the make command will also need adjusting to suit.

DuckyPolice, 2024-05-14:
ubuntu@ubuntu:~/projects/PicoBB/console/pico$ make BOARD=vgaboard_sd CYW43=NONE STDIO=USB SERIAL_DEV=0 GRAPH=Y SOUND=PWM
cd ../../src/pico && symbols/header_funcs.py /home/ubuntu/projects/PicoBB/console/pico//build/symbols_sdk.txt /home/ubuntu/projects/PicoBB/console/pico//build/pico_stub.c \
	../../src/pico/symbols/sdk_headers.txt ../../src/pico/symbols/sdk_stdio_usb.txt
../../src/pico/symbols/merge_funcs.py build/sympico.h build/symbols_sdk.txt
rm -rf build/picobb.ld
cd build && PICO_SDK_PATH=/home/ubuntu/pico-sdk \
	cmake -DPICO_BOARD=vgaboard_sd -DSTDIO=USB -DLFS=Y -DFAT=N -DSOUND=PWM \
		-DSTACK_CHECK=4 -DMIN_STACK=Y -DPRINTER= -DSERIAL_DEV=0 \
		-DCYW43=NONE -DBBC_SRC=../../BBCSDL -DGRAPH=Y -DOS_RAM= -DUSB_CON=0 \
		--no-warn-unused-cli -S ../../../src/pico -B .
Not searching for unused variables given on the command line.
Using PICO_SDK_PATH from environment ('/home/ubuntu/pico-sdk')
PICO_SDK_PATH is /home/ubuntu/pico-sdk
Defaulting PICO_PLATFORM to rp2040 since not specified.
Defaulting PICO platform compiler to pico_arm_gcc since not specified.
-- Defaulting build type to 'Release' since not specified.
PICO compiler is pico_arm_gcc
-- Include PICO-EXTRAS
CMake Error at CMakeLists.txt:19 (include):
  include could not find load file:

    /external/pico_extras_import.cmake

-- The C compiler identification is GNU 10.3.1
-- The CXX compiler identification is GNU 10.3.1
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/arm-none-eabi-gcc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/arm-none-eabi-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/arm-none-eabi-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
Build type is Release
PICO target board is vgaboard_sd.
Using board configuration from /home/ubuntu/projects/PicoBB/boards/vgaboard_sd.h
-- Found Python3: /usr/bin/python3 (found version "3.10.12") found components: Interpreter 
TinyUSB available at /home/ubuntu/pico-sdk/lib/tinyusb/src/portable/raspberrypi/rp2040; enabling build support for USB.
BTstack available at /home/ubuntu/pico-sdk/lib/btstack
cyw43-driver available at /home/ubuntu/pico-sdk/lib/cyw43-driver
Pico W Bluetooth build support available.
lwIP available at /home/ubuntu/pico-sdk/lib/lwip
Pico W Wi-Fi build support available.
mbedtls available at /home/ubuntu/pico-sdk/lib/mbedtls
-- BBC BASIC Upstream source from ../../BBCSDL
-- Using upstream expression evaluation code with REDUCE_STACK_SIZE
-- Using memory protection to detect stack overrun
-- No Pico W support
-- Generate sound using PWM
-- Including LFS filesystem for flash storage
-- BBC Basic console I/O will be on USB
-- Framebuffer Graphics Support
-- Link with space for VGA video
-- Configuring incomplete, errors occurred!
See also "/home/ubuntu/projects/PicoBB/console/pico/build/CMakeFiles/CMakeOutput.log".
See also "/home/ubuntu/projects/PicoBB/console/pico/build/CMakeFiles/CMakeError.log".
make: *** [../../src/pico/Makefile:176: build/Makefile] Error 1

DuckyPolice, 2024-05-14:
and in cmakelists.txt if i remove the if statement after `include($ENV{PICO_SDK_PATH}/external/pico_sdk_import.cmake)`
i can get it to start building but end up hitting
`/home/ubuntu/projects/PicoBB/src/pico_gui/picofbuf.c:23:10: fatal error: pico/scanvideo.h: No such file or directory
   23 | #include "pico/scanvideo.h"`

Memotech-Bill, 2024-05-14:
VGA output requires the scanvideo routines from [pico-extras](https://github.com/raspberrypi/pico-extras). Install this at the same level as **pico-sdk** and **PicoBB**.

You also need to set the environment variable PICO_EXTRAS_PATH to the path of the **pico-extras** folder.

If you had installed the SDK using:

````
wget https://raw.githubusercontent.com/raspberrypi/pico-setup/master/pico_setup.sh
````

then this would already have been installed and the environment variables set.

DuckyPolice, 2024-05-14:
ok. i did a bit of digging and found pico-extras as one of your repos. when i set an env variable after building that, i got a successful picobb build! you should add pico-extras as a dependency in the readme because that was very hard to find out.

Memotech-Bill, 2024-05-14:
You should take pico-extras from the official Raspberry Pi source: https://github.com/raspberrypi/pico-extras (as per my earlier link). The fork in my GitHub account is out of date.

Had you installed the Pico SDK using the **pico-setup.sh** script, as per [Getting Started with Raspberry Pi Pico](https://datasheets.raspberrypi.com/pico/getting-started-with-pico.pdf), then **pico-extras** would have been installed automatically.

DuckyPolice, 2024-05-16:
ok i got my vga board and it is working perfectly! quick question though, what is the wav file that picowav.bbc wants? where can i download it? and is there any online tools that would allow me to manipulate a wav file into the needed bitrates and stuff for picowav to accept it?

DuckyPolice, 2024-05-16:
(oops accidentily closed it. after this question gets answered i will close it)

closed 2024-05-19 by DuckyPolice