Memotech-Bill/PicoBB #25: graphics display broken

DuckyPolice, 2024-05-19:
it outputs vga fine but makes the sprites look like tv static. also how do i build a VGA version but with usb serial. i dont like having to constantly run `*output 15` every time i make a typo. also, a clockspeed star command would be nice because some programs run too quickly.

Memotech-Bill, 2024-05-19:
> it outputs vga fine but makes the sprites look like tv static.

BBC BASIC does not have sprites

> also how do i build a VGA version but with usb serial.

As per earlier issue or the README file

> i dont like having to constantly run `*output 15` every time i make a typo.

So include the star command in the program

> also, a clockspeed star command would be nice because some programs run too quickly.

Use the BBC BASIC SYS command to call the C SDK routine to adjust the clock frequency

DuckyPolice, 2024-05-20:
for sprits i mean custom tiles or whatever they are supposed to be in games. custom graphics. it doesn't render correctly. for the building thing, i meant like not console version but basically output 14 all the time where picocom is only for keypresses but picocom doesn't actually get anything returning such as text.basically a com port serial keyboard (when i say com port serial i mean picocom or regular com port stuff as a computer as a host.)

Memotech-Bill, 2024-05-20:
> for sprits i mean custom tiles or whatever they are supposed to be in games. custom graphics. it doesn't render correctly.

I still don't know what you mean. Give a short example of the code that you think does not work.

> for the building thing, i meant like not console version but basically output 14 all the time where picocom is only for keypresses but picocom doesn't actually get anything returning such as [...]

That is too specialist a requirement for me to implement. You will have to edit the source code yourself. I will advise that the *output command sets the value of `optval`, and that the output goes through routine `xeqvdu` in src/picocli.c. Edit that routine to call `fbufvdu` for VGA output or `org_xeqvdu` for console output.

DuckyPolice, 2024-05-20:
[image]
what it should look like

[image]
what it looks like for unknown reasons

DuckyPolice, 2024-05-20:
pretty much any game can find on https://www.bbcmicro.co.uk/ has that issue. also half of them have syntax errors or other things that shouldn't be happening. so maybe work on functionality and compatability? also, it would be difficult but running disassembly would be awesome for games becase so many games use it

DuckyPolice, 2024-05-20:
(i recreated what it looks like with pixel art but it looks pretty much the same as the graphics issue)

Memotech-Bill, 2024-05-20:
Until I know what BBC BASIC command you are executing, which is not having the desired effect, I cannot investigate.