Memotech-Bill/PicoBB #27: Running 'lanchat' causes 'patience' to crash

rtrussell, 2024-10-20:
This seems consistently to cause **patience.bbc** to crash with a Hard Fault:
```
Start BBC BASIC on the Pico W.
Run wifi_cfg.bbc to set up the network connection.
Run lanchat.bbc (runs correctly).
Run patience.bbc (crashes).
```
Substituting **snake.bbc** for **patience.bbc** has the same issue, but both work without crashing if **lanchat.bbc** isn't run first.

Memotech-Bill, 2024-10-22:
This may be resolved, but I am not sure why.

There was a problem with running a program that loaded libraries after running a program that used **socklib**. The was related to changing HIMEM. That moved the memory protection block that is supposed to detect stack overruns. When HIMEM was moved up, the memory protection block was placed immediately above HIMEM. When a subsequent program attempts to load a library, the scan for library already loaded runs into the protected memory area causing a hard fault.

The code has now been revised so that providing HIMEM remains at or below `libase` then the protected area remains above `libtop`. The protected area is now only moved to HIMEM if HIMEM is moved above `libase`.

That would have caused many programs to hard fault if run after **lanchat** (or any other program using **socklib**).

I quickly tried running **snake** after **lanchat** and it appeared to run correctly. However **snake** appears not to include any libraries, so I am not sure why this seems to be fixed.

rtrussell, 2024-10-22:
You can expect BBC BASIC to read from (but not write to) memory **immediately above** HIMEM because to discover whether it needs to 'pop' a record off the stack it checks for the presence of a sentinel value.  It would be a significant overhead, in both code size and time, to have to check for the stack pointer being >= HIMEM in addition to this sentinel test.

This is generally safe because libraries are loaded above HIMEM so there must be some valid memory there (the assumption is that the first 4 or 8 bytes of the first library won't be mistaken for a sentinel value, I'm not sure whether that's guaranteed but it's statistically vanishingly unlikely).

If HIMEM is moved down to create space for a buffer etc. this should also be safe, so long as the first few bytes of that buffer are readable and don't contain one of the sentinel values.  But if the memory immediately above HIMEM is protected and not readable, then definitely BBC BASIC will crash when it looks for the sentinel.

To be completely safe you would need to leave a gap of at least one pointer size (4 or 8 bytes) immediately above HIMEM unused (but readable).  This probably isn't documented anywhere explicitly, but is a common feature of all my BASICs.

Memotech-Bill, 2024-10-22:
The start of the protected region is 32-byte aligned, at least 64 bytes above **HIMEM** or **libtop**. See [here](https://github.com/Memotech-Bill/PicoBB/blob/0b4b577c0aa08a4427da9be001470bbc8da13eb1/src/bbpico.c#L1792).