Memotech-Bill/PicoBB #29: How to set STDIO on UART with no USB

gertk64, 2024-10-28:
I am trying to create a PicoBB system with VGA, STDIO over UART and no USB but it gives me  the following error during make with the following options:

`~/projects/Raspberry_Pi_Pico/pico/PicoBB/console/pico$ make clean && make BOARD=vgaboard_cut CYW43=NONE STDIO=UART SERIAL_DEV=1 GRAPH=Y SOUND=PWM PRINTER=N`

`/home/gert/projects/Raspberry_Pi_Pico/pico/pico-sdk/lib/tinyusb/src/common/tusb_common.h:74:10: fatal error: tusb_option.h: No such file or directory`

This message seems to note that it still tries to use some USB stuff.

The board I use is a Pimoroni pico demo VGA with the GPIO20 and GPIO21 connections cut. These GPIO pins I would like to use for a serial terminal/keyboard and use VGA for main output. 

When I do a make with STDIO=USB+UART it compiles without errors but after uploading nothing happens or responds on the UART pins (20/21). Trying *output 0 just outputs to the USB connection again.

Any ideas?

Memotech-Bill, 2024-10-28:
That is a configuration I have not tested. It will take a while to investigate.

If you want to use VGA as main output, and a keyboard as input, what is wrong with the GUI build? Keyboard attached directly to the Pico USB port.

Is there any reason for avoiding the USB port?

gertk64, 2024-10-28:
Well the reason is that I have reverse engineered an Motorola Alphamate with a very nice keyboard (and a single line LCD) so it works as a RS232 terminal and would like to attach this to a Pico running BBCbasic and have VGA output. Since in the makefile you can select USB and/or UART I assumed it would be easy to just select UART.
Thanks for your time.

Memotech-Bill, 2024-10-29:
OK, I have pushed a couple of minor fixes. Pull my changes and then build with:

````
make clean && make BOARD=vgaboard_cut CYW43=NONE STDIO=UART SERIAL_DEV=0 GRAPH=Y SOUND=PWM PRINTER=N
````

Note that if the serial connection is used for STDIO, then it is not available as a serial device, hence `SERIAL_DEV=0`.

gertk64, 2024-10-29:
> OK, I have pushed a couple of minor fixes. Pull my changes and then build with:
>
> ```
> make clean && make BOARD=vgaboard_cut CYW43=NONE STDIO=UART SERIAL_DEV=0 GRAPH=Y SOUND=PWM PRINTER=N
> ```
>
> Note [...]

This seems to work, thanks!

Now figuring out to set the default output to VGA while keeping the input from UART aka doing an *output 14