Memotech-Bill/PicoBB #30: Build error: CYW43_PIN_WL_SDIO_1 undeclared

rtrussell, 2025-03-01:
Following exactly the same build steps that I've used successfully before, I'm now getting the following error:

`/home/pi/pico/pico-sdk/lib/cyw43-driver/src/cyw43_ctrl.c:282:31: error: 'CYW43_PIN_WL_SDIO_1' undeclared
`

Do I need to update the Pico-SDK to fix this?  I would have tried that, but it's such a slow process that I thought I'd ask before potentially wasting my time if that's not the issue.

Memotech-Bill, 2025-03-01:
I have never seen that error message so I don't know.

I don't know what your build method is.

My build method has changed, as documented in the [README.md](https://github.com/Memotech-Bill/PicoBB) file.

The **console/pico_w** (and **bin/pico_w**) folder no longer exists. Instead change to the **console/pico** folder, and do one of:

````
make BOARD=pico_w
````
or
````
make BOARD=pico2_w
````

I am using the latest version of the SDK. Updating the SDK should not be slow or difficult. Use the following commands:

````
cd pico-sdk
git pull
git submodule update
````

Memotech-Bill, 2025-03-02:
I think **./pico_setup.sh** is doing a full re-install rather than an upgrade.

Perhaps the biggest limitation of Raspberry Pi OS is that they don't provide an upgrade path from one major release to the next. There are plenty of complaints about that on the forum.

On my x86 laptop I run Ubuntu. That will happily do an in situ upgrade from one long tern support release to the next. I have never tried Ubuntu on RPi, but I would guess that it supports in situ upgrades on the RPi as well.

One way of reducing the amount of effort in performing an upgrade is to mount /home on a flash drive, rather than the SD card containing the OS. Then you don't have to copy your own files. It is still necessary to re-install all the software you are using, although there are third party tools that can automate that.

rtrussell, 2025-03-02:
> I think **./pico_setup.sh** is doing a full re-install rather than an upgrade.

Probably, but isn't a 'full re-install' safer anyway?  Either way, if  the upgrade requires **cmake 3.17** the same problem would arise.

> Perhaps the biggest limitation of Raspberry Pi OS is that they don't provide an upgrade path from one major release to the next. There are plenty of complaints about that on the forum.

Indeed.  Every other flavour of Linux I have used provides such an upgrade path.  I tried editing the sources list from _buster_ to _bullseye_ and then doing a **dist-upgrade**, but not surprisingly it didn't work.

Something I've noticed is that whenever I build BBC BASIC for the Pico it is downloading and re-building **picotool** at the same time, which is time-consuming.  I'm surprised that the dependencies deem that to be necessary, is there some way of preventing it?

Memotech-Bill, 2025-03-02:
Build and install picotool

````
git clone https://github.com/raspberrypi/picotool.git
cd picotool
mkdir build
cd build
cmake ..
make
sudo make install
````

It is also worthwhile installing pioasm

````
cd pico-sdk/tools/pioasm
mkdir build
cd build
cmake ..
make
sudo make install
````

It is necessary to re-make and re-install these every time the SDK is updated.

closed 2025-09-14 by Memotech-Bill