Memotech-Bill/PicoBB #31: Changes required to bbpico.c

rtrussell, 2025-03-01:
Because of a Pull Request by Jonathan Harston you will need to add definitions for the `getmodeno()` and `setext()` functions to **bbpico.c** because they have now - entirely reasonably - been moved out of the core interpreter into a customisable module.  You can take suitable definitions from `BBCSDL/src/bbccon.c`, or if you can do something more useful with them please do!

Memotech-Bill, 2025-03-02:
I think I can do something useful with `setext()` at least for FAT filesystem.

What is it supposed to do with the file pointer?

* Leave it at its previous position.
* Position it at the new end of file.
* Position it at the beginning of file.

Also, is there any defined behaviour for negative arguments?

* Error.
* Truncate to zero length.
* Truncate the specified number of bytes from the end of the file.

For Flash filesystem, I don't think pre-allocation of space makes any sense, but it is still possible to truncate files.

rtrussell, 2025-03-02:
> What is it supposed to do with the file pointer?

Good question, I can't see anything said about that in the RISC OS BBC BASIC Manual, which is the most 'official' reference we have.

Experimentally,  and unhelpfully, _Brandy BASIC_ and _BBC BASIC for Windows_ behave differently: Brandy leaves PTR# **unchanged** (unless it was beyond the new end of the file) but BB4W always moves it to the new **end of the file**.

> Also, is there any defined behaviour for negative arguments?

Again, I can only determine that by experiment.  Brandy reports an **error** message, BB4W seems to **ignore** the attempt to change the length.

So there's no consistency, and therefore the programmer can't make any assumption either way and you can probably do whatever you want!  :-)

Memotech-Bill, 2025-03-02:
I have now pushed the required changes.

For FAT on the SD card, `setext()` will either truncate or expand a file. If the file is initially zero in size, then the allocated space will be contiguous.

For LittleFS on the flash memory, `setext()` will truncate a file if required. It will silently ignore a request to expand a file.

rtrussell, 2025-04-18:
It looks like you are using `getmode()` not `getmodeno()`.  Unfortunately `getmode()` clashes with a standard shared library [function](https://linux.die.net/man/3/getmode) of the same name.  It's possible that this isn't an issue with the Pico build, but it is with some others.

Memotech-Bill, 2025-04-18:
Pulled the latest version of BBCSDL, then changed `getmode()` to `getmodeno()` in **picobb.c**.

Update committed.